Nicholas3388/LuaNode #2: Is this project on track to completion?

carterw, 2016-06-14:
I see that it didn't reach the minimum goal on kickstarter. Will you be implementing Lua and sending out the development kits people signed up for?

Nicholas3388, 2016-06-17:
Hi carterw,
No matter the kickstarter campaign is successful or not, we will keep updating Luanode and providing Esp32 dev kit. The Esp32 kit will go retail after the campaign closed. The kit is being producing, once it goes retail you can purchase it via our website: www.doit.am. Thanks for paying attention to this project.

Best regards

carterw, 2016-06-20:
Thank you for the info, I am glad to hear that the project is moving forward. Do you have any estimate on when the Esp32 development kit will be available?

Nicholas3388, 2016-06-24:
Hi carterw,
We estimate the kit will be available for retail in the middle of July. we still need some time for hardware producing, sorry for the delay. I will inform you once it is ready for sale. Thanks.

Best regards

Nicholas3388, 2016-06-25:
Hi carterw,
I got some news from Espressif, they will start providing Esp32 chip in August, and then we will start producing Esp32 development kit. Maybe before the end of August, the kit will be available for retail.

Best regards

Nicholas3388, 2016-09-26:
Hi @carterw,
Sorry for delay, we are busy testing the dev-kit these days. Post some screenshots here: [http://ww3.sinaimg.cn/mw690/999babe3gw1f8745dne7zj20zk0qo44r.jpg](url)  [http://ww1.sinaimg.cn/mw690/999babe3gw1f8745hvyr9j20zk0qodo2.jpg](url)  We plan to sell it next month. Thanks for your attention to our product.

rzr, 2017-11-07:
It think this ussie can be closed, since it's available on various different places

I bought mine at:

https://www.banggood.com/ESP32-Development-Board-WiFiBluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board-p-1109512.html?p=P514146481305201609J#&cur_warehouse=CN

and I am wondering how to spot fake ones, if any