Nicholas3388/LuaNode #5: SPI and PWM

vslinuxdotnet, 2017-02-07:
Hello Nicholas,

I now porting my project from ESP8266 to EST32 and notice that at the moment I can't work with PWM and SPI.
I activate the modules in main/include/user_modules.h and compiled, then I flash the ESP32, in console terminal show the libs loaded in boot, all ok so far.
But testing with LUA scripts, the PWM and SPI run without no results, and also no errors.

**LUA PWM test code:**

```
print("Test buzzer ESP32!")
gpio.mode(23,2)
pwm.setup(23, 440, 512)  
pwm.start(23)  
tmr.delay(500 * 300) 
pwm.stop(23) 
```

**SPI detection teste code:**

```
id=0 
sda=21
scl=22

function read_reg(dev_addr, reg_addr)
     i2c.start(id)
     i2c.address(id, dev_addr ,i2c.TRANSMITTER)
     i2c.write(id,reg_addr)
     i2c.stop(id)
     i2c.start(id)
     i2c.address(id, dev_addr,i2c.RECEIVER)
     c=i2c.read(id,1)
     i2c.stop(id)
     return c
end

function find_dev(i2c_id, dev_addr)
     i2c.start(i2c_id)
     c=i2c.address(i2c_id, dev_addr ,i2c.TRANSMITTER)
     i2c.stop(i2c_id)
     return c
end

print("Scanning I2C Bus...")
i2c.setup(id,sda,scl,i2c.SLOW)
for i=0,127 do
     if find_dev(id, i)==true then
     print("Device found at address 0x"..string.format("%02X",i))
     end
end
print("Done!")
```

any help, comment about ?
Thank You

Nicholas3388, 2017-02-08:
@vslinuxdotnet Hi, there are still have some problems with PWM and i2c. I'm working on it. I'll fix it and respond to you as soon as possible.

Nicholas3388, 2017-02-09:
@vslinuxdotnet  Hi, I just fix bugs in PWM. Now you can update and test PWM module.

The following methods are provide by PWM module:
1. pwm.setup(pin, freqency, duty, channel)
      pin: pin to set
      frequency: frequency to set
      duty: duty to set. range from 0 to 8192
      channel: used channel, valid channel range from 0 to 7
2. pwm.stop(channel)
3. pwm.getduty(channel)
4. pwm.setduty(channel, duty)
5. pwm.getfreq()
6. pwm.setfreq(frequency)

Note: once pwm.setup is called, your device start to generate PWM.

The following example is to make the blue LED on DOIT ESP32 dev-board blink per seconds. There is a blue LED connect to GPIO2 on DOIT ESP32 dev-board.

examples:
pwm.setup(2, 1, 4096, 0);  -- blink per seconds
duty=pwm.getduty(0);
print(duty);
freq = pwm.getfreq();
print(freq);
tmr.delay(10);
pwm.setfreq(2);  -- blink twice per seconds
tmr.delay(10);
pwm.stop(0);

I will fix the problem with I2C tomorrow;)

vslinuxdotnet, 2017-02-09:
Hello there!!! it works fine, with led or buzzer, PWM fixed :)
God Job Nicholas.
I wait for the I2C 👍

Nicholas3388, 2017-02-11:
@vslinuxdotnet  Hi, I solve the problem, Please update the project to test i2c
the i2c module provide the following methods:
1. i2c.setup(mode, i2c_num, scl_pin, sda_pin, addr)
    mode: can be i2c.SLAVE/i2c.MASTER
    i2c_num: can be i2c.I2C_0/i2c.I2C_1 (two i2c on esp32)
    scl_pin: the scl pin
    sda_pin: the sda pin
    addr: the addr range from 1 to 128
2. i2c.write(mode, i2c_num, addr, content)
    content: the content can be string, number
3. i2c.read(mode, i2c_num, addr, len)

example:
-- since there are two i2c on esp32, I setup one i2c as master and the other one as slave, and read/write them on the esp32
-- slave :
--    GPIO25 is assigned as the data signal of i2c slave port
--    GPIO26 is assigned as the clock signal of i2c slave port
-- master:
--    GPIO18 is assigned as the data signal of i2c master port
--    GPIO19 is assigned as the clock signal of i2c master port
-- connect GPIO18 with GPIO25
-- connect GPIO19 with GPIO26

i2c.setup(i2c.SLAVE, i2c.I2C_0, 26, 25, 40)
i2c.setup(i2c.MASTER, i2c.I2C_1, 19, 18)
-- slave write something, master read
i2c.write(i2c.SLAVE, i2c.I2C_0, 40, "helloworld")
c=i2c.read(i2c.MASTER, i2c.I2C_1, 40, 10)
print(c)

tmr.delay(5)
-- master write something, slave read
i2c.write(i2c.MASTER, i2c.I2C_1, 40, "test")
c=i2c.read(i2c.SLAVE, i2c.I2C_0, 40, 4)
print(c)

vslinuxdotnet, 2017-02-11:
Hi @Nicholas3388 once again "touche", it works.
In my case only need the master because ESP32 is always the Master and the other devices was Slaves.
So I test your example and works fine...
I build a Script in Lua to scan for addresses available in the I2C bus.
If you what, you can add it inside Lua examples, in your project :)

My code below:
```
--Lua I2c bus scanner - 11/02/2017
 
sda=18
scl=19

function find_dev(dev_addr)
     c=i2c.read(i2c.MASTER, i2c.I2C_1, dev_addr, 1)
     if c == "" then
        return false
     else
       return true
     end
end

i2c.setup(i2c.MASTER, i2c.I2C_1, scl, sda)
print("Scanning I2C Bus...")
for i=0,127 do
     if find_dev(i)==true then
        print("Device found at address 0x"..string.format("%02X",i).."!")
     end
end
print("Done!")
```

Thank you

Nicholas3388, 2017-02-12:
@vslinuxdotnet   OK, Thanks

closed 2017-02-13 by Nicholas3388