Nicholas3388/LuaNode #6: Run init.lua or init.lc at startup

vslinuxdotnet, 2017-02-13:
Hello @Nicholas3388 ,

I change the "components/lua/lua.c" to run the startup lua script in boot.

Is not at 100%, but it works.
Check the diff:
```
@@ -20,7 +20,9 @@
 #include "my_uart.h"
 #include "rom/uart.h"
 #include "rom/ets_sys.h"
+#include <spiffs.h>

+static int dofsfile (lua_State *L, const char *name);
 char line_buffer[LUA_MAXINPUT];

 static lua_State *globalL = NULL;
@@ -213,45 +215,45 @@
   lua_remove(L, 1);  /* remove line */
   return status;
 }
-#if 0
+//#if 0
 //doit
 static void mydofile(lua_State *L)
 {
   char *fn="init.lua";
   int file_fd=-1;
-  lua_spiffs_mount();
+  //lua_spiffs_mount();

-  file_fd = SPIFFS_open(&fs,fn,SPIFFS_RDONLY,0);
+  file_fd = luaL_loadfsfile(L,fn);
   if(file_fd<0)
   {
     char *fn2 = "init.lc";
-    file_fd = SPIFFS_open(&fs,fn2,SPIFFS_RDONLY,0);
+    file_fd = luaL_loadfsfile(L,fn2);
     if(file_fd<0)
       l_message(NULL,"cannot open init.lua or init.lc");
     else
     {
-      SPIFFS_close(&fs,file_fd);
-      dofile(L,"init.lc");
+      //SPIFFS_close(&fs,file_fd);
+      dofsfile(L,"init.lc");
     }
   }
   else
   {
-     SPIFFS_close(&fs,file_fd);
-     dofile(L,"init.lua");
+     //SPIFFS_close(&fs,file_fd);
+     dofsfile(L,"init.lua");
   }
 }

 static void dotty (lua_State *L) {
   int status;
-  const char *oldprogname = progname;
+  //const char *oldprogname = progname;

-  //mydofile(L);//doit
-
+  mydofile(L);//doit
+/*
   progname = NULL;
   while ((status = loadline(L)) != -1) {
     if (status == 0) status = docall(L, 0, 0);
     report(L, status);
-    if (status == 0 && lua_gettop(L) > 0) {  /* any result to print? */
+    if (status == 0 && lua_gettop(L) > 0) {  // any result to print?
       lua_getglobal(L, "print");
       lua_insert(L, 1);
       if (lua_pcall(L, lua_gettop(L)-1, 0, 0) != 0)
@@ -260,13 +262,13 @@
                                lua_tostring(L, -1)));
     }
     lua_gc(L, LUA_GCCOLLECT, 0);//doit
-  }
+  }*/
   lua_settop(L, 0);  /* clear stack */

   luai_writeline();//doit
   /*fputs("\n", stdout);
   fflush(stdout);*/
-  progname = oldprogname;
+  //progname = oldprogname;
 }

@@ -278,7 +280,7 @@
   fname = argv[n];
   if (strcmp(fname, "-") == 0 && strcmp(argv[n-1], "--") != 0)
     fname = NULL;  /* stdin */
-  status = luaL_loadfile(L, fname);
+  status = luaL_loadfsfile(L, fname);
   lua_insert(L, -(narg+1));
   if (status == 0)
     status = docall(L, narg, 0);
@@ -286,7 +288,7 @@
     lua_pop(L, narg);
   return report(L, status);
 }
-#endif
+//#endif

 /* check that argument has no extra characters at the end */
 #define notail(x)      {if ((x)[2] != '\0') return -1;}
@@ -414,7 +416,7 @@
   //if (has_v) print_version();
   s->status = runargs(L, argv, (script > 0) ? script : s->argc);
   if (s->status != 0) return 0;
-#if 0
+//#if 0
   if (script)
     s->status = handle_script(L, argv, script);
   if (s->status != 0) return 0;
@@ -425,9 +427,9 @@
       print_version();
       dotty(L);
     }
-    else dofile(L, NULL);  /* executes stdin as a file */
+    else dofsfile(L, NULL);  /* executes stdin as a file */
   }
-#endif
+//#endif
   return 0;
 }

```

Nicholas3388, 2017-02-15:
@vslinuxdotnet  I'll figure it out as soon as possible :)

Nicholas3388, 2017-02-19:
@vslinuxdotnet  Hi, the init.lua file will be running after Lua is initialised, now. Please update LuaNode to test.

closed 2017-02-22 by Nicholas3388