Nicholas3388/LuaNode #8: Board Dimension and Pinouts

pixma, 2017-02-25:
I would like to have the development kit's dimension and pin position in mm unit if possible.
I want to make some shields for the development board quickly, which will help me to be more accurate about the project i am working on.

So, where can i find the dimensions, pin positions, and pin mappings for the esp32 development board?

pixma, 2017-02-25:
There are many such dev boards out in the market so, Particularly, I am looking for DOIT ESP32 board dimensions, pin positions measurements. Because that's what i have bought.

Nicholas3388, 2017-02-27:
@pixma  Hi, you can download the DOIT ESP32 board dimention from the following link: https://www.dropbox.com/s/0ivvd85s46u1fgn/ESP32_dimension.png?dl=0

closed 2017-03-03 by Nicholas3388

pixma, 2017-04-03:
Link : http://pixma.github.io 

Scroll down a bit

Capacitive Touch Sensor Shield 
=========

for DOIT dev Board. Tested Cap Touch Buttons and the performance. Amazing to work with ESP32.
I have revised the board with more proper cap touch pinouts. If anybody need those shields please let me know, will send you - 

inbox me : pixma38@gmail.com

Image Reference:

[image]

Nicholas3388, 2017-04-04:
@pixma  Great job! glad to hear news from you;)