Nicholas3388/LuaNode #9: conflicting types for 'ledc_timer_config'

zorglub76, 2017-04-22:
When building LuaNode32, I'm getting this error:

CC pwm.o
CC periph_ctrl.o
CC ledc.o
/home/mare/esp32/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./ledc.c:217:11: error: conflicting types for 'ledc_timer_config'
 esp_err_t ledc_timer_config(ledc_timer_config_t* timer_conf)
           ^
In file included from /home/mare/esp32/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./ledc.c:22:0:
/home/mare/esp32/esp-idf/components/driver/include/driver/ledc.h:131:11: note: previous declaration of 'ledc_timer_config' was here
 esp_err_t ledc_timer_config(const ledc_timer_config_t* timer_conf);
           ^
/home/mare/esp32/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./ledc.c:259:11: error: conflicting types for 'ledc_channel_config'
 esp_err_t ledc_channel_config(ledc_channel_config_t* ledc_conf)
           ^
In file included from /home/mare/esp32/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./ledc.c:22:0:
/home/mare/esp32/esp-idf/components/driver/include/driver/ledc.h:118:11: note: previous declaration of 'ledc_channel_config' was here
 esp_err_t ledc_channel_config(const ledc_channel_config_t* ledc_conf);
           ^
/home/mare/esp32/esp-idf/make/component_wrapper.mk:176: recipe for target 'ledc.o' failed
make[1]: *** [ledc.o] Error 1
/home/mare/esp32/esp-idf/make/project.mk:386: recipe for target 'mydriver-build' failed
make: *** [mydriver-build] Error 2

Nicholas3388, 2017-04-24:
@zorglub76  Hi, I build LuaNode few minutes ago. It build without problem. I'm doing it on Ubuntu 14.04, with the newest esp-idf and the official prebuild-toolchain download from the link: http://esp-idf.readthedocs.io/en/latest/linux-setup.html .  Is your esp-idf too ancient? Maybe you can update your esp-idf, as well as toolchain.

galleguayo, 2017-06-30:
@Nicholas3388 Hi, I also have similar compilation errors.
`
/home/luisperez/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use O
                                      ^
/home/luisperez/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:211:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((addr), READ_PERI_REG);                                
             ^
/home/luisperez/iotbits_esp32/esp-idf/components/soc/esp32/include/soc/soc.h:230:36: note: in expansion of macro 'READ_PERI_REG'
             WRITE_PERI_REG((reg), (READ_PERI_REG(reg)|(mask)));                        
                                    ^
/home/luisperez/iotbits_esp32/esp-idf/examples/LuaNode/LuaNode32/components/mydriver/./periph_ctrl.c:164:13: note: in expansion of macro 'SET_PERI_REG_MASK'
             SET_PERI_REG_MASK(DPORT_PERIP_RST_EN_REG, DPORT_UHCI1_RST);
             ^
make[1]: *** [periph_ctrl.o] Error 1
make: *** [mydriver-build] Error 2
`
I am using the latest version of toolchain: 
https://dl.espressif.com/dl/xtensa-esp32-elf-linux32-1.22.0-61-gab8375a-5.2.0.tar.gz

Please can you help me?

Best regards

Nicholas3388, 2017-07-03:
@galleguayo  Hi, I build LuaNode without problem. Can you try to `make clean` the project, and then build again. Is there any other info output? The content you listed is not enough for debugging