Nicholas3388/LuaNode #10: node.restart() kills the USB connection

erknie, 2017-04-23:
I noticed that if you call node.restart()  the boot process kills the USB ->serial connection to the computer you are using to talk to Luanode. I am using a Mac running Yosemite, and tried it on several different terminal programs. Zterm, miniterm.py, picocom, they all hang when you issue an node.restart() so you have to quit the terminal program and restart it.

I am using the device  /dev/cu.SLAB_USBtoUART to talk to LuaNode on the DOIT ESP-32 Dev board which comes with LuaNode already installed.

Nicholas3388, 2017-04-24:
@erknie Hi, I am using /dev/tty.SLAB_USBtoUART on Mac OS Sierra to talk to LuaNode. I used `screen /dev/tty.SLAB_USBtoUART 115200` to input commands. It works well, when input node.restart(), the device will restart without killing USB connection. Maybe you can try to download and install my CP210x driver from the following link: https://pan.baidu.com/s/1pL4TPfP , and then use `screen`  command to talk to LuaNode.

erknie, 2017-04-26:
Ok, I did some more tests and it works in Arduino IDE serial monitor, on an older Mac OSX so you may be right. I will try your driver tonight.

rzr, 2017-11-07:
it works fine on my debian linux system running GNU screen