Nicholas3388/LuaNode #11: Can't get broad work

anhtuanmanga, 2017-04-26:
i bought a broad from this link
https://world.taobao.com/item/545443841643.htm?spm=a312a.7700714.0.0.M5KZlH#detail
and i flashed firmware from branch LuaNode_1.2.0 and blue and red led on broad work but when i open port in ESPlorer only Communication with MCU.. show and i can't upload code to ESP32
can you help me pls and sorry for my abd english

Nicholas3388, 2017-04-27:
@anhtuanmanga Hi, please use the master branch. The latest sources are pushed to the master  branch. The other branches still have some problems.

anhtuanmanga, 2017-04-27:
i tried with firmware from master branch too
here is my setup
[image]
but after finish flash only red led work
do i have to press any button when flash firmware?

Nicholas3388, 2017-04-27:
@anhtuanmanga  The FLASH SIZE is 32Mbit

anhtuanmanga, 2017-04-27:
still not work
do you have any document or guidline?

Nicholas3388, 2017-04-27:
@anhtuanmanga Hi, we've published a book to introduce programming with ESP32, but it is wrote in Chinese. If you are not Chinese, I'm afraid you won't need it. 

For the firmware we provided in LuaNode, it will output some libs loaded information when power up. The following screenshot is the output info from my Macbook. [http://img1.ph.126.net/MDPdhBuNuTJoh7tqzS8AGg==/6632360791373571970.png](url)

Can you show me what's your output info when power up? You can see the output from ESPlorer, note that the baud rate is 115200.

anhtuanmanga, 2017-04-28:
thank you for the help.
I have press RST button on ESPlorer, now the console show log like your screenshot.

anhtuanmanga, 2017-05-11:
can you explaine for me about boot and en button

Nicholas3388, 2017-05-12:
press en button will trigger hard reset. If you fail to connect to ESP32 device while using `flash_download_tools` to download firmware, press boot button will re-synchronize with ESP32.

anhtuanmanga, 2017-05-12:
So they don't connect to GPIO1 and GPIO0?

Nicholas3388, 2017-05-20:
The boot button connect to GPIO0, you can see that from the schematic. The schematic download link post on LuaNode github page

closed 2017-05-20 by anhtuanmanga