Nicholas3388/LuaNode #12: ESP-32 compiler error

mandoyj, 2017-04-27:
implicit declaration of function 'SPIParamCfg' [-Werror=implicit-function-declaration]
   SPIParamCfg (
   ^

Nicholas3388, 2017-04-27:
@mandoyj Hi, this function is defined in esp-idf/components/esp32/include/rom/spi_flash.h. Maybe you forgot to include the head file like this `#include "rom/spi_flash.h"`

mandoyj, 2017-04-27:
@Nicholas3388 Thank you for reply. The function 'SPIParamCfg' has been removed.
which commit do you use?

Nicholas3388, 2017-05-03:
@mandoyj  Hi, I think you should use `esp_rom_spiflash_config_param` instead of `SPIParamCfg` for ESP32

0203Gaurav, 2017-09-09:
hi.
i am getting two error. please help me out

1. 
gaurav@gaurav-Aspire-E5-573:~/Desktop/esp123/hello_world/main$ make menuconfig
Makefile:8: /home/gaurav/bin:/home/gaurav/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/gaurav/esp/xtensa-esp32-elf/bin:/gaurav/esp/esp-idf/make/project.mk: No such file or directory
make: *** No rule to make target `/home/gaurav/bin:/home/gaurav/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/gaurav/esp/xtensa-esp32-elf/bin:/gaurav/esp/esp-idf/make/project.mk'.  Stop.

2.
gaurav@gaurav-Aspire-E5-573:~/Desktop/esp123/hello_world/main$ gcc -o hello hello_world_main.c
hello_world_main.c:10:31: fatal error: freertos/FreeRTOS.h: No such file or directory
compilation terminated.

Nicholas3388, 2017-09-11:
@0203Gaurav  Hi,
For the first issue, maybe you didn't export IDF_PATH, export as follow: export IDF_PATH=/your_path. For the second issue, did you setup compile environment as the following link: http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html 

If you have setup, you just need to execute make in the directory of hello_world_main.c