Nicholas3388/LuaNode #13: where is the LuaNode-Code-Reference?

StefanL38, 2017-04-27:
Hello,

maybe I got tomatoes on my eyes. I was not able to find a full LuaNode code-reference.
I tried some nodeMCU-commands built in LuaLoader. But they resulted in errors like

"wifi.setmode(wifi.STATION) wifi.sta.getap(function(t) if t then print("\n\nVisible Access Points:\n") for k,v in pairs(t) do l = string.format("%-10s",k) print(l.."  "..v) end else print("Try again") end  end)

stdin:1: attempt to call field 'getap' (a nil value)

so please give me a hint where I can find the code-reference

best regards  

Stefan

Nicholas3388, 2017-05-01:
@StefanL38 Hi, there is no 'getap' method in wifi module. There are difference between NodeMCU commands and LuaNode. But I'll try to use the same commands as NodeMCU in the future. There is a document within `LuaNode_Esp32` directory, the doc describe methods within Lua modules. But the doc is ancient, there are already some new module and methods. I'll try to update the doc asap.

StefanL38, 2017-05-01:
Hello Nicholas,

thank you ver ymuch for answering.
I can understand if you have other things to do with a higher priority.
Anyway if your product should become a great success I think it would be good to have a complete documentation. What I'm missing is the commands for interacting with all kinds of sensors.
I²C, 1-Wire, SPI, and preconfigured bitbanging for DHT11 humidity-sensors, LED-drivers like WS2812,
PWM, RC-Servo-signal-creation ADC, MQTT, BME280 etc., etc, etc.
Take a look at the nodeMCU-customize firmware creater at https://nodemcu-build.com/

If LuaNode shall be a REAL improvement in my opinion it has to support ALL the things nodeMCU does AND add something new. 
From the documentation that is provided yet some users might come to the conclusion LuaNode just supports
5.2.1 gpio	7
5.2.2 node	8
5.2.3 tmr	9
5.2.4 file	10
5.2.5 wifi	13
5.2.6 net	15
5.2.7 thread	17
5.2.8 lpeg	19
5.2.9 utils	20

and all the other things are missing, because there is no dosumentation about them.

Nicholas3388, 2017-05-04:
@StefanL38  Yes, you're right. There is still tons of work to do for LuaNode, I'll add the sensor drivers when I'm free. I'll do that as soon as possible, once I finish my current project, which is a wifi & BLE switcher. I have to focus on the project, right now. Anyway, thanks for your advice.