Nicholas3388/LuaNode #14: ESP-32 compiling on Mac OSX 10.10.5

erknie, 2017-04-29:
I am getting a lot of errors trying to compile the current Luanode source. I am using the current git version, and the current git version of esp-idf, and the toolchain binaries xtensa-esp32-elf-osx-1.22.0-61-gab8375a-5.2.0.tar.gz

Getting several errors like:

`CC ledc.o
/Volumes/esp32/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./ledc.c:259:11: error: conflicting types for 'ledc_channel_config'
 esp_err_t ledc_channel_config(ledc_channel_config_t* ledc_conf)
           ^
In file included from /Volumes/esp32/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./ledc.c:22:0:
/Volumes/esp32/esp-idf/components/driver/include/driver/ledc.h:118:11: note: previous declaration of 'ledc_channel_config' was here
 esp_err_t ledc_channel_config(const ledc_channel_config_t* ledc_conf);
           ^
make[1]: *** [ledc.o] Error 1
make: *** [mydriver-build] Error 2
`

I fix those by adding the word const to esp_err_t ledc_channel_config(ledc_channel_config_t* ledc_conf) so it matches the version in esp-idf and the compilation proceeds.

I am currently stuck on this error and have to ask for help:

`CC flash_api.o
/Volumes/esp32/LuaNode/LuaNode_Esp32/LuaNode32/components/platform/./flash_api.c: In function 'update_flash_chip_size':
/Volumes/esp32/LuaNode/LuaNode_Esp32/LuaNode32/components/platform/./flash_api.c:34:20: error: 'esp_rom_spiflash_chip_t {aka struct <anonymous>}' has no member named 'deviceId'
     g_rom_flashchip.deviceId,
                    ^
make[1]: *** [flash_api.o] Error 1
make: *** [platform-build] Error 2`

art-mx, 2017-04-30:
+1 On this. Compiling on a fresh Ubuntu 14.04. Getting similar "conflicting types" error, as well as multiple warnings but for gpio.c... 
`/home/artyom/.local/share/Trash/files/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./gpio.c: At top level:
/home/artyom/.local/share/Trash/files/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./gpio.c:264:11: error: conflicting types for 'gpio_config'
 esp_err_t gpio_config(gpio_config_t *pGPIOConfig)
           ^
In file included from /home/artyom/.local/share/Trash/files/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./gpio.c:19:0:
/home/artyom/esp/esp-idf/components/driver/include/driver/gpio.h:224:11: note: previous declaration of 'gpio_config' was here
 esp_err_t gpio_config(const gpio_config_t *pGPIOConfig);
           ^
/home/artyom/.local/share/Trash/files/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./gpio.c: In function 'gpio_config':
/home/artyom/.local/share/Trash/files/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./gpio.c:306:17: warning: 'PIN_PULLUP_EN' is deprecated [-Wdeprecated-declarations]
                 PIN_PULLUP_EN(io_reg);
                 ^
In file included from /home/artyom/esp/esp-idf/components/driver/include/driver/gpio.h:22:0,
                 from /home/artyom/.local/share/Trash/files/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./gpio.c:19:
/home/artyom/esp/esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:99:49: note: declared here
 static inline void __attribute__ ((deprecated)) PIN_PULLUP_EN(uint32_t PIN_NAME)
                                                 ^
/home/artyom/.local/share/Trash/files/LuaNode/LuaNode_Esp32/LuaNode32/components/mydriver/./gpio.c:308:17: warning: 'PIN_PULLUP_DIS' is deprecated [-Wdeprecated-declarations]
                 PIN_PULLUP_DIS(io_reg);
                 ^
`
Please fix asap!

Nicholas3388, 2017-05-04:
@erknie @art-mx  Hi, Fixed it, please update to test again. Thanks.

erknie, 2017-05-05:
Yep, that compiles for me now. Thanks.

closed 2017-05-05 by erknie