Nicholas3388/LuaNode #16: Build Problem: Nothing to be done for FORCE

TobySchultze, 2017-07-01:
I'm trying to build for my ESP32. I've followed your wiki entry and everything worked great. But when I try to build the project with `./gen_misc.sh` I'm getting a response `make: Nothing to be done for FORCE `.
I've edited the thre variables as mentioned. The error occures when I try to build the default `wifilister` or `alexa_multi_esp32` app.
Thanks for your help.

Nicholas3388, 2017-07-03:
@TobySchultze  Hi, it is not necessary to run `gen_misc.sh`, just enter the `alexa_multi_esp32` directory and then run `make` on the terminal to build the project. Make sure to export your IDF_PATH before building. For more details on environment setup, visit link: http://esp-idf.readthedocs.io/en/latest/get-started/index.html#setup-toolchain

TobySchultze, 2017-07-04:
Great! Thank you so much @Nicholas3388 ! The official doc helped a lot. Maybe you could update your documentation for other users.

closed 2017-07-04 by TobySchultze