Nicholas3388/LuaNode #19: Error: conflicting types for "pthread_mutexattr_t"

fakstory, 2017-09-13:
Hi, I am able to build with the esp_idf, my setup is correct on 2 machines, but I got this error on both. I am missing something?

`In file included from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:18:0:
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/pthread/include/pthreadx.h:70:35: error: conflicting types for 'pthread_mutexattr_t'
 typedef struct pthread_mutex_attr pthread_mutexattr_t;
                                   ^
In file included from /home/fak/esp/esp-idf/components/newlib/include/stdio.h:48:0,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:2:
/home/fak/esp/esp-idf/components/newlib/include/sys/types.h:457:3: note: previous declaration of 'pthread_mutexattr_t' was here
 } pthread_mutexattr_t;
   ^
In file included from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:18:0:
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/pthread/include/pthreadx.h:80:22: error: conflicting types for 'pthread_condattr_t'
 typedef unsigned int pthread_condattr_t;
                      ^
In file included from /home/fak/esp/esp-idf/components/newlib/include/stdio.h:48:0,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:2:
/home/fak/esp/esp-idf/components/newlib/include/sys/types.h:469:3: note: previous declaration of 'pthread_condattr_t' was here
 } pthread_condattr_t;         /* a condition attribute object */
   ^
In file included from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:18:0:
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/pthread/include/pthreadx.h:81:13: error: conflicting types for 'pthread_t'
 typedef int pthread_t;
             ^
In file included from /home/fak/esp/esp-idf/components/newlib/include/stdio.h:48:0,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:2:
/home/fak/esp/esp-idf/components/newlib/include/sys/types.h:308:20: note: previous declaration of 'pthread_t' was here
 typedef __uint32_t pthread_t;            /* identify a thread */
                    ^
In file included from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:18:0:
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/pthread/include/pthreadx.h:82:13: error: conflicting types for 'pthread_key_t'
 typedef int pthread_key_t;
             ^
In file included from /home/fak/esp/esp-idf/components/newlib/include/stdio.h:48:0,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:2:
/home/fak/esp/esp-idf/components/newlib/include/sys/types.h:473:20: note: previous declaration of 'pthread_key_t' was here
 typedef __uint32_t pthread_key_t;        /* thread-specific data keys */
                    ^
In file included from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:18:0:
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/pthread/include/pthreadx.h:88:29: error: conflicting types for 'pthread_cond_t'
 typedef struct pthread_cond pthread_cond_t;
                             ^
In file included from /home/fak/esp/esp-idf/components/newlib/include/stdio.h:48:0,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:2:
/home/fak/esp/esp-idf/components/newlib/include/sys/types.h:462:20: note: previous declaration of 'pthread_cond_t' was here
 typedef __uint32_t pthread_cond_t;       /* identify a condition variable */
                    ^
In file included from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:18:0:
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/pthread/include/pthreadx.h:94:29: error: conflicting types for 'pthread_once_t'
 typedef struct pthread_once pthread_once_t;
                             ^
In file included from /home/fak/esp/esp-idf/components/newlib/include/stdio.h:48:0,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:2:
/home/fak/esp/esp-idf/components/newlib/include/sys/types.h:478:3: note: previous declaration of 'pthread_once_t' was here
 } pthread_once_t;       /* dynamic package initialization */
   ^
In file included from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:18:0:
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/pthread/include/pthreadx.h:129:29: error: conflicting types for 'pthread_attr_t'
 typedef struct pthread_attr pthread_attr_t;
                             ^
In file included from /home/fak/esp/esp-idf/components/newlib/include/stdio.h:48:0,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:2:
/home/fak/esp/esp-idf/components/newlib/include/sys/types.h:363:3: note: previous declaration of 'pthread_attr_t' was here
 } pthread_attr_t;
   ^
In file included from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:18:0:
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/pthread/include/pthreadx.h:134:0: warning: "PTHREAD_CREATE_DETACHED" redefined
 #define PTHREAD_CREATE_DETACHED 1
 ^
In file included from /home/fak/esp/esp-idf/components/newlib/include/stdio.h:48:0,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:2:
/home/fak/esp/esp-idf/components/newlib/include/sys/types.h:322:0: note: this is the location of the previous definition
 #define PTHREAD_CREATE_DETACHED 0
 ^
In file included from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:18:0:
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/pthread/include/pthreadx.h:135:0: warning: "PTHREAD_CREATE_JOINABLE" redefined
 #define PTHREAD_CREATE_JOINABLE 2
 ^
In file included from /home/fak/esp/esp-idf/components/newlib/include/stdio.h:48:0,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:2:
/home/fak/esp/esp-idf/components/newlib/include/sys/types.h:323:0: note: this is the location of the previous definition
 #define PTHREAD_CREATE_JOINABLE  1
 ^
In file included from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:18:0:
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/pthread/include/pthreadx.h:189:0: warning: "PTHREAD_MUTEX_NORMAL" redefined
 #define PTHREAD_MUTEX_NORMAL        1
 ^
In file included from /home/fak/esp/esp-idf/components/newlib/include/stdio.h:48:0,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:2:
/home/fak/esp/esp-idf/components/newlib/include/sys/types.h:399:0: note: this is the location of the previous definition
 #define PTHREAD_MUTEX_NORMAL     0
 ^
In file included from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:18:0:
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/pthread/include/pthreadx.h:191:0: warning: "PTHREAD_MUTEX_RECURSIVE" redefined
 #define PTHREAD_MUTEX_RECURSIVE     3
 ^
In file included from /home/fak/esp/esp-idf/components/newlib/include/stdio.h:48:0,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:2:
/home/fak/esp/esp-idf/components/newlib/include/sys/types.h:411:0: note: this is the location of the previous definition
 #define PTHREAD_MUTEX_RECURSIVE  1
 ^
In file included from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:18:0:
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/pthread/include/pthreadx.h:192:0: warning: "PTHREAD_MUTEX_DEFAULT" redefined
 #define PTHREAD_MUTEX_DEFAULT       4
 ^
In file included from /home/fak/esp/esp-idf/components/newlib/include/stdio.h:48:0,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:2:
/home/fak/esp/esp-idf/components/newlib/include/sys/types.h:430:0: note: this is the location of the previous definition
 #define PTHREAD_MUTEX_DEFAULT    3
 ^
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c: In function 'app_main':
/home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:109:3: warning: 'system_restart' is deprecated [-Wdeprecated-declarations]
   system_restart();
   ^
In file included from /home/fak/esp/esp-idf/components/freertos/include/freertos/portable.h:126:0,
                 from /home/fak/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
                 from /home/fak/esp/LuaNode/LuaNode_Esp32/LuaNode32/main/./app_main.c:4:
/home/fak/esp/esp-idf/components/esp32/include/esp_system.h:88:6: note: declared here
 void system_restart(void) __attribute__ ((deprecated, noreturn));
      ^
make[1]: *** [/home/fak/esp/esp-idf/make/component_wrapper.mk:239: app_main.o] Error 1
make: *** [/home/fak/esp/esp-idf/make/project.mk:421: component-main-build] Error 2
`

fakstory, 2017-09-14:
What is the git commit hash of the ESP_IDF are you using?

Nicholas3388, 2017-09-20:
@fakstory  Hi, sorry for delay. I used the latest commit. The conflicting error is fixed

fakstory, 2017-09-21:
That's what I thought, thanks, I'm closing this.

closed 2017-09-21 by fakstory