Nicholas3388/LuaNode #20: Building firmware compiler error: unknown type name 'os_timer_t'

axonometric, 2017-10-25:
/home/build/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/./wifi.c:78:8: error: unknown type name 'os_timer_t'
 static os_timer_t wifi_sta_status_timer;
        ^
Linux mike-VirtualBox 4.10.0-37-generic #41~16.04.1-Ubuntu SMP Fri Oct 6 22:42:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

dl8bb, 2017-11-05:
Hi folks,
  any solution or hints for this issue? I run into the same problem.

Kind regards,
birko
--

jeffsauer, 2017-11-10:
When compiling the WiFi module, the make command doesn't appear to be finding the esp_timer.h file located in LuaNode32/main/include.  So, my temporary solution was to simply:

cd LuaNode32
cp -rp main/include/* components/modules

jeffsauer, 2017-11-10:
When compiling using the esp-idf v2.1, I was getting an error saying that nvs_open() failed. Looks like one needs to first initialize NVS now, apparently in previous version the WiFi init did that, but no longer. So, I edited main.c and added the following include:

#include "nvs_flash.h"

and then inside of initialise_wifi() I added:

nvs_flash_init();

just before the call to tcpip_adapter_init();