Nicholas3388/LuaNode #22: schematic of esp32 doit board

sarwadenj, 2017-10-31:
Hi @Nicholas3388 ,

Please share or add schematic pdf in repository. It will be helpful.

Thanks.

george-vieira, 2017-12-07:
I just bought this off https://www.amazon.com/gp/product/B0754DT83W/ but the picture and the device I received don't match up.
[image]
As you can see from my photo, this is a different revision.
Can you please add the proper pinouts on the website somewhere?

Nicholas3388, 2018-01-05:
@sarwadenj  Hi, sorry for delay. I'm busy last month. You can download the schematic from the following link: [http://bbs.doit.am/forum.php?mod=attachment&aid=NjE4fDRkZWZiYWFhfDE1MTUxNDExOTR8MHw0MzI%3D](url)

george-vieira, 2018-01-06:
404. it's found here
https://www.dropbox.com/s/jefwxxtufgwg0ex/esp32_Schematic%20Prints.pdf?dl=0

george-vieira, 2018-01-06:
https://www.dropbox.com/s/jefwxxtufgwg0ex/esp32_Schematic%20Prints.pdf?dl=0

skypigr, 2022-07-12:
None of the above links works at the time of writting, any updates?

skypigr, 2022-07-12:
> None of the above links works at the time of writting, any updates?

I found one that matches my board (30PIN): https://drive.google.com/file/d/1uNpxMy-Hqx4FZSb3FGyQaJGmOtt1SZDU/view?usp=sharing