Nicholas3388/LuaNode #24: Build Error on Windows - MSYS32

sonuame, 2017-12-25:
I have configured the full build tool chain on windows and make is working quite okay, but not able to build LuaNode firmware.

here it is the log

`/d/GitClones/EspIDF/make/component_common.mk:1: Deprecated feature: No longer necessary to include component_common.mk from /d/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/main/component.mk
CC build/modules/tmr.o
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c:84:2: error: unknown type name 'os_timer_t'
  os_timer_t os;
  ^
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c:99:8: error: unknown type name 'os_timer_t'
 static os_timer_t rtc_timer;
        ^
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c: In function 'tmr_now':
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c:157:2: warning: 'system_get_time' is deprecated [-Wdeprecated-declarations]
  uint32_t now = 0x7FFFFFFF & system_get_time();
  ^
In file included from D:/GitClones/EspIDF/components/freertos/include/freertos/portable.h:126:0,
                 from D:/GitClones/EspIDF/components/freertos/include/freertos/FreeRTOS.h:105,
                 from D:/GitClones/EspIDF/components/driver/include/driver/uart.h:28,
                 from D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/platform/include/platform.h:9,
                 from D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c:54:
D:/GitClones/EspIDF/components/esp32/include/esp_system.h:96:10: note: declared here
 uint32_t system_get_time(void)  __attribute__ ((deprecated));
          ^
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c: In function 'tmr_register':
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c:179:3: warning: implicit declaration of function 'os_timer_disarm' [-Wimplicit-function-declaration]
   os_timer_disarm(&tmr->os);
   ^
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c:187:2: warning: implicit declaration of function 'os_timer_setfn' [-Wimplicit-function-declaration]
  os_timer_setfn(&tmr->os, alarm_timer_common, (void*)id);
  ^
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c: In function 'tmr_start':
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c:201:3: warning: implicit declaration of function 'os_timer_arm' [-Wimplicit-function-declaration]
   os_timer_arm(&tmr->os, tmr->interval, tmr->mode==TIMER_MODE_AUTO);
   ^
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c: In function 'rtc_callback':
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c:316:4: warning: 'system_restart' is deprecated [-Wdeprecated-declarations]
    system_restart();
    ^
In file included from D:/GitClones/EspIDF/components/freertos/include/freertos/portable.h:126:0,
                 from D:/GitClones/EspIDF/components/freertos/include/freertos/FreeRTOS.h:105,
                 from D:/GitClones/EspIDF/components/driver/include/driver/uart.h:28,
                 from D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/platform/include/platform.h:9,
                 from D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c:54:
D:/GitClones/EspIDF/components/esp32/include/esp_system.h:88:6: note: declared here
 void system_restart(void) __attribute__ ((deprecated, noreturn));
      ^
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c: At top level:
D:/GitClones/LuaNode/LuaNode_Esp32/LuaNode32/components/modules/tmr.c:99:19: warning: 'rtc_timer' defined but not used [-Wunused-variable]
 static os_timer_t rtc_timer;
                   ^
make[1]: *** [/d/GitClones/EspIDF/make/component_wrapper.mk:274: tmr.o] Error 1
make: *** [d:/GitClones/EspIDF/make/project.mk:449: component-modules-build] Error 2
`

Please Help me out

Nicholas3388, 2018-01-05:
@sonuame  Hi, sorry for delay, I'm busy last month. Bugs fixed now.

sonuame, 2018-01-05:
Freaked out again on os_timer_t
check the attached log
[log.txt](https://github.com/Nicholas3388/LuaNode/files/1606324/log.txt)

Nicholas3388, 2018-01-10:
@sonuame  Hi, please update your esp-idf firstly, and then update LuaNode by `git pull origin master`, finally, `make` LuaNode project. I think it will be OK.