Nicholas3388/LuaNode #28: Build failed with linker conflicts: dhcpserver of espconn vs lwip

buergi, 2018-03-14:
The build failed for me due to linker conflicts of the dhcpserver of espconn and lwip.

I started with a fresh build environment, as described here http://esp-idf.readthedocs.io/en/latest/get-started/windows-setup.html, cloned esp-idf and LuaNode, exported the necessary PATH and IDF_PATH envvars and executed make in LuaNode32.
Here are the error messages I got

```
C:/msys32/home/user/esp/LuaNode/LuaNode_Esp32/LuaNode32/build/lwip\liblwip.a(dhcpserver.o): In function `node_remove_from_list':
C:/msys32/home/user/esp/esp-idf/components/lwip/apps/dhcpserver.c:246: multiple definition of `node_remove_from_list'
C:/msys32/home/user/esp/LuaNode/LuaNode_Esp32/LuaNode32/build/espconn\libespconn.a(dhcpserver.o):C:/msys32/home/user/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/espconn/dhcpserver.c:97: first defined here
C:/msys32/home/user/esp/LuaNode/LuaNode_Esp32/LuaNode32/build/lwip\liblwip.a(dhcpserver.o): In function `dhcps_start':
dhcpserver.c:(.text.dhcps_start+0x0): multiple definition of `dhcps_start'
C:/msys32/home/user/esp/LuaNode/LuaNode_Esp32/LuaNode32/build/espconn\libespconn.a(dhcpserver.o):dhcpserver.c:(.text.dhcps_start+0x0): first defined here
C:/msys32/home/user/esp/LuaNode/LuaNode_Esp32/LuaNode32/build/lwip\liblwip.a(dhcpserver.o): In function `dhcps_stop':
dhcpserver.c:(.text.dhcps_stop+0x0): multiple definition of `dhcps_stop'
C:/msys32/home/user/esp/LuaNode/LuaNode_Esp32/LuaNode32/build/espconn\libespconn.a(dhcpserver.o):dhcpserver.c:(.text.dhcps_stop+0x0): first defined here
C:/msys32/home/user/esp/LuaNode/LuaNode_Esp32/LuaNode32/build/lwip\liblwip.a(dhcpserver.o): In function `dhcps_coarse_tmr':
dhcpserver.c:(.text.dhcps_coarse_tmr+0x0): multiple definition of `dhcps_coarse_tmr'
C:/msys32/home/user/esp/LuaNode/LuaNode_Esp32/LuaNode32/build/espconn\libespconn.a(dhcpserver.o):dhcpserver.c:(.text.dhcps_coarse_tmr+0x0): first defined here
C:/msys32/home/user/esp/LuaNode/LuaNode_Esp32/LuaNode32/build/espconn\libespconn.a(dhcpserver.o):(.literal.parse_msg+0x14): undefined reference to `wifi_softap_set_station_info'
C:/msys32/home/user/esp/LuaNode/LuaNode_Esp32/LuaNode32/build/espconn\libespconn.a(dhcpserver.o): In function `parse_msg':
C:/msys32/home/user/esp/LuaNode/LuaNode_Esp32/LuaNode32/components/espconn/dhcpserver.c:1045: undefined reference to `wifi_softap_set_station_info'
collect2.exe: error: ld returned 1 exit status
```

Any solution to fix this?

seurat-atreides, 2018-03-19:
I'm having the same issue.
I'm building on Ubuntu 16.04 and have installed the tool chain from Espressif: xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0 and the esp-idf according to the instructions found in the "ESP-IDF Programming / Get Started" guide.

seurat-atreides, 2018-03-20:
I got the code to build by deleting file: dhcpserver.c, found under LuaNode32/components/espconn/ .
This code seems to be outdated and the current release of esp-idf has a working version under components/lwip/apps .
I will test the build tonight!

seurat-atreides, 2018-03-21:
It worked like expected.