Nicholas3388/LuaNode #30: KiCAD/EgaleCad footprint?

proffalken, 2018-05-09:
Hello,

I've just been redirected here by the Amazon seller of an ESP32 board (https://www.amazon.co.uk/gp/product/B071JR9WS9/ ).  I've no idea whether they've just copied your design wholesale or whether it is an official product, but if you have any existing CAD footprints that you are able to share, that would be very much appreciated.

KiCAD is preferred, however EAGLE would also suffice.

Driver55, 2018-05-18:
I also need the same documentation.
moreover it would be good to uniquely identify the R15 resistor to exclude the blue LED from GPIO_02 to allow the use of the SDMMC interface

anlpereira, 2018-06-15:
I would like the 3D file. I have done Eagle footprint. If anyone needs just email me anlpereira@gmail.com

VictorLamoine, 2020-08-26:
Here is a KiCAD version:
[image]

https://gitlab.com/VictorLamoine/kicad_esp32_devkit_v1

I have not used it yet so be sure to check it. Many datasheets are wrong so it's quite a headache to get things right.