Nicholas3388/LuaNode #33: Using ESP32 Devkit 1 with Arduino IDE

m-menon, 2018-10-07:
Is there any documentation on how to get ESP32 Devkit 1 with Arduino IDE? I brought a few of these boards and not been able to program it using Arduino IDE. I cloned the esp32 repo from github. After I connect the board to my computer and press the "BOOT" button on the board, I can see the following in the Arduino IDE Serial Monitor

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2488
load:0x40078000,len:7128
load:0x40080000,len:252
entry 0x40080034
[0;32mI (1100) cpu_start: Pro cpu up.[0m
[0;32mI (1100) cpu_start: Single core mode[0m
[0;32mI (1101) heap_alloc_caps: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1114) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM[0m
[0;32mI (1135) heap_alloc_caps: At 3FFB8160 len 00027EA0 (159 KiB): DRAM[0m
[0;32mI (1156) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1177) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1199) heap_alloc_caps: At 40093EA4 len 0000C15C (48 KiB): IRAM[0m
[0;32mI (1220) cpu_start: Pro cpu start user code[0m
[0;32mI (1279) cpu_start: Starting scheduler on PRO CPU.[0m
Mounting flash filesystem...
mount res: 0, 0

=======================================
LuaNode: https://github.com/Nicholas3388/LuaNode
Version: 1.2.2
=======================================

load lib: base
load lib: package
load lib: table
load lib: string
load lib: gpio
load lib: pwm
load lib: node
load lib: wifi
I (1554) wifi: wifi firmware version: 224c254
I (1558) wifi: config NVS flash: enabled
I (1561) wifi: config nano formating: disabled
I (1569) wifi: Init dynamic tx buffer num: 32
I (1570) wifi: Init dynamic rx buffer num: 64
I (1574) wifi: wifi driver task: 3ffc0668, prio:23, stack:3584
I (1579) wifi: Init static rx buffer num: 10
I (1583) wifi: Init dynamic rx buffer num: 0
I (1587) wifi: Init rx ampdu len mblock:7
I (1591) wifi: Init lldesc rx ampdu entry mblock:4
I (1595) wifi: wifi power manager task: 0x3ffc5a20 prio: 21 stack: 2560
load lib: file
load lib: tmr
load lib: i2c
load lib: uart
load lib: utils
load lib: lpeg
load lib: net
load lib: thread
load lib: mqtt
load lib: nvs
other libs:
lua: cannot open init.lua

> 
```

But when I try to program I get an error "An error occured while trying to upload the sketch"

Can anyone point me to the documentation on how to use this board with Arduino IDE? Thanks

aloveric, 2020-08-10:
Same issue.
Did you solve it?

m-menon, 2020-08-10:
It's been a while so I don't recollect exactly but do you want to take a look at [this](https://github.com/espressif/arduino-esp32/issues/544) in the meantime? 

I'll try to locate my board and if I do then I may be able to comment further.

aloveric, 2020-08-11:
I am also using Arduino IDE. If you want to upload the code to the board, just need to press boot button when shows up "connecting ___...____ ". The upload will be started

Tatuck, 2020-11-05:
Hello! did you solve it?

chenyuhan1997, 2022-04-23:
hey,i found the same problem,i flash the FIRMWARE, and then i can't use it with arduino,i feel so worried,and i can't find any way solve it!!!!!

chenyuhan1997, 2022-04-23:
oh,man,i filish my test,and if you refresh your FirmWare, i think you can wait some time and test a example just like "WiFi SCAN",and then if you find it work ,i think it is success

SpiderDKing, 2022-08-15:
在arduino出现connecting....的时候要长按开发板上的boot键